OpenSprinkler/OpenSprinkler-Firmware #346: why's OSPi being so slow on my rpi3b?

wommy, 2025-05-07:
ive been trying to debug this for a week now

fresh install of latest raspberrypi 64 lite

did an `sudo apt update ; sudo apt upgrade -y`
came back nothing to do

nothing else installed

``` bash
sudo apt install -y git 
git clone --recurse-submodules https://github.com/OpenSprinkler/OpenSprinkler-Firmware.git 
cd OpenSprinkler-Firmware
sudo ./build.sh ospi
```

answered y
rebooted

it takes like 5 minutes to load the device password page

submit button doesnt click, but if i wait another 5 mins i eventually hit the home page

even tried a `sudo rpi-update`

curl is showing an extremely long response time of 4.5 mins

``` bash

root@hostname ~# time curl 192.168.10.35:8080
<!DOCTYPE html><html><head><meta name="viewport" content="width=device-width,initial-scale=1.0,minimum-scale=1.0,user-scalable=no"></head><body><script>var ver=221,ipas=0;</script><script src="https://ui.opensprinkler.com/js/home.js"></script></body></html>
________________________________________________________
Executed in  269.49 secs      fish           external
   usr time   12.66 millis  251.00 micros   12.41 millis
   sys time   12.54 millis  119.00 micros   12.42 millis

```

what do you guys need from me?

rayshobby, 2025-05-07:
The firmware has been tested on RPi 0 which is much weaker than 3b and is still running pretty fast. The homepage does require loading UI assets from ui.opensprinkler.com so if your Internet is slow calmer that could explain why it takes so long. But you can use the opensprinkler mobile app, or you can also host the UIS locally, to avoid internet connection issues

wommy, 2025-05-08:
I've tried

how do I host locally

wommy, 2025-05-08:
also I'm pretty sure
aka it started happening when
the web socket lib got introduced

wommy, 2025-05-08:
also i actually cannot use the app because i keep getting this unable to connect, because i guess the request is timing out because it takes 5 minutes

rayshobby, 2025-05-08:
I honestly don't know. I haven never seen this issue before. You said you installed the "latest raspberrypi 64 lite" -- maybe report the specific version? So we can see if it's reproducible.

rayshobby, 2025-05-08:
Look, the websocket is unrelated -- it's only used when you enable OTC token for remote access. Like I said, I've never seen this issue you reported. We've tested the latest OpenSprinkler firmware on multiple RPis and have never seen this issue. Our demo website:
[demo.opensprinkler.com](demo.opensprinkler.com)
runs the latest firmware on Debian. So I don't think this is a common problem. Have you tried it on a different RPi, just in case maybe the issue is due to the RPi hardware itself?

wommy, 2025-05-08:
websockets cause alot of network noise, i worked extensively with them in the fall building a media streaming app
long polling isnt really needed outside of chatrooms, Server Sent Events were a much more minimal protocol
websockets are easy but they come with alot of baggage, i spent months replacing them with equiv SSE
my switches RSTP and IGMP i think specifically blocks that kind of stuff is why i mentioned it

i have tried switching Pis but ill try again
i have a Zero2b and a 4

the Zero2b doesnt have soldered headers, do they need to be to interface or would just snug fit

rayshobby, 2025-05-08:
Are you using OTC for remote access? If not, like I said, WebSockets are not activated at all in the firmware. Unless if you believe just linking to the WebSockets library would cause the issue (which I don't believe is possible), then it's unrelated.

At the moment, you are just trying to figure out what's causing the delay, so whether the Rpi has headers or not doesn't matter -- you don't even need to plug it into OSPi -- just power it with a USB cable, and test the software. The network stuff has nothing to do with whether RPi is connected to OSPi or not.

wommy, 2025-05-08:
pi4 works fine even over wifi,
the zero2w wont even boot, im going to try to flash the 32bit lite

wommy, 2025-05-08:
i am starting to think its hardware,
i cannot get the zero to boot to save my life, even with 32bit
and the 3b is having the same issues even w 32bit

ill run to the store and grab another zero

rayshobby, 2025-05-08:
Just to be clear: by "pi4 works fine" did you mean it runs the firmware fine and you can now access the firmware? Or did you only mean that it boots fine?

For RPi 0, I would recommend that you stay with Bullseye -- while it theoretically works with Bookworm (32-bit) the performance will suffer. It should work ok with Bullseye.

wommy, 2025-05-08:
everything worked on the pi4, booted and firmware ran normal

100% aging hardware, i got a new Zero2w everything worked fine first try
my rpi3b is dated 2016, and my zero 2017, they mostly sat in a drawer 

ive lost too many braincells on this already but ill probably take a crack at the bullseye install
just for completion

thanks for your patience

closed 2025-05-08 by wommy

wommy, 2025-05-08:
100% bookworm and 64 bit

just did a bullseye 32 bit and it all just worked, no lag

on a Zero w not a Zero 2w

im guessing if i swap it into my 3b itll run perfectly fine too

rayshobby, 2025-05-08:
OK, that's good to know. Thanks for the update.