OpenSprinkler/OpenSprinkler-Firmware #348: Fail to build container image

varac, 2025-05-09:
I fail to create the container image following the README_Docker.md, using Podman (although I'm pretty sure that this is not the problem). I tried both latest release (`221(1)`) and current HEAD, both with the same result. 

* OsPi, Debian bookworm
* Podman V4.3.1

```
$ podman build -t opensprinkler .
...
[2/3] STEP 8/8: RUN make VERSION=${BUILD_VERSION}                                                                
g++ -std=gnu++14 -D -DSMTP_OPENSSL -Wall -include string.h -include cstdint -Iexternal/TinyWebsockets/tiny_websoc
kets_lib/include -Iexternal/OpenThings-Framework-Firmware-Library/   -c -o main.o main.cpp
<command-line>: error: macro names must be identifiers                                                           
main.cpp: In function 'void do_loop()':                 
main.cpp:694:39: warning: comparison of integer expressions of different signedness: 'time_os_t' {aka 'long int'}
 and 'uint32_t' {aka 'unsigned int'} [-Wsign-compare]   
  694 |                         if (curr_time >= os.nvdata.rd_stop_time) {  // rain delay is over
      |                             ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~                                          
main.cpp:698:52: warning: comparison of integer expressions of different signedness: 'uint32_t' {aka 'unsigned in
t'} and 'time_os_t' {aka 'long int'} [-Wsign-compare]                                                            
  698 |                         if (os.nvdata.rd_stop_time > curr_time) {  // rain delay starts now
main.cpp:1022:57: warning: comparison of integer expressions of different signedness: 'time_os_t' {aka 'long int'
} and 'uint32_t' {aka 'unsigned int'} [-Wsign-compare]  
 1022 |                 if (os.status.safe_reboot && (curr_time > reboot_timer)) {                               
      |                                               ~~~~~~~~~~^~~~~~~~~~~~~~                                   
main.cpp:1039:54: warning: comparison of integer expressions of different signedness: 'time_os_t' {aka 'long int'
} and 'uint32_t' {aka 'unsigned int'} [-Wsign-compare]  
 1039 |                 } else if(reboot_timer && (curr_time > reboot_timer)) {                                  
      |                                            ~~~~~~~~~~^~~~~~~~~~~~~~                                      
In file included from OpenSprinkler.h:67,               
                 from main.cpp:27:                      
rpitime.h: At global scope:                             
rpitime.h:26:12: warning: 'int month(time_t)' defined but not used [-Wunused-function]                           
   26 | static int month(time_t ct) {                                                                            
      |            ^~~~~                                
rpitime.h:21:12: warning: 'int day(time_t)' defined but not used [-Wunused-function]                             
   21 | static int day(time_t ct) {                     
      |            ^~~                                  
rpitime.h:16:12: warning: 'int second(time_t)' defined but not used [-Wunused-function]                          
   16 | static int second(time_t ct) {                  
      |            ^~~~~~                               
rpitime.h:11:12: warning: 'int minute(time_t)' defined but not used [-Wunused-function]                          
   11 | static int minute(time_t ct) {                  
      |            ^~~~~~                               
rpitime.h:6:12: warning: 'int hour(time_t)' defined but not used [-Wunused-function]                             
    6 | static int hour(time_t ct) {                    
      |            ^~~~                                 
make: *** [<builtin>: main.o] Error 1                   
Error: building at STEP "RUN make VERSION=${BUILD_VERSION}": while running runtime: exit status 2
```

Any idea how to fix this ?

mellodev, 2025-05-09:
Hello @varac I just gave it a try on my Debian machine using the build commands listed in the readme and it succeeded:

[image]

I ran your error message through AI and it suggests the empty `-D` argument is causing the issue you see:

[image]

Looking at the Makefile, it appears that should define the version. Perhaps try removing that or setting the value manually?

[image]

varac, 2025-05-09:
Thanks ! I gave it a try, it didn't work, but then I decided to go with the packages containers of this project, which is working fine.

closed 2025-05-09 by varac