OpenSprinkler/OpenSprinkler-Firmware #350: Feature Request: Provide rain information via MQTT

Vrumfondel, 2025-05-12:
Since I found the online-weather information of current rain status very unreliable, I would like to provide the information if it is currently raining - and trigger the delay from this - via MQTT (I got a rain sensor in my home automation).

Would be nice if this can be added as an option to online or wired sensor.

rayshobby, 2025-05-14:
The features added by this PR (#340) should support this, but we are currently still reviewing the PR and trying to iron out all potential issues.

Vrumfondel, 2025-05-14:
In the change description for this PR I only see the ability to OUTPUT the weatherdata via MQTT, but not INPUT it.
Do I miss something there?

rayshobby, 2025-05-21:
I believe Stefan's firmware (PR #340) can read weather data from MQTT as well. I will double check.