OpenSprinkler/OpenSprinkler-Firmware #352: Feature Request: cycle and soak

Wetzel402, 2025-05-14:
Cycle and Soak is a great feature included on some commercial irrigation controllers that promotes deeper root growth and reduces runoff. Currently opensprinkler requires manual configuration using the "repeat every" setting which is unintuitive and confusing. I would like to suggest a simple checkbox option to enable cycle and soak. It could either be calculated automatically or have additional configuration options that become available when the box is checked.

rayshobby, 2025-05-19:
I am not sure how the current 'repeat every' is unintuitive and confusing -- let's say you want to cycle and soak 5 zones, each 10 minutes at a time, and you want every zone to be watered for a total of 30 minutes, then isn't it just setting each zone to water for 10 minutes, repeat every 50 minutes, for 2 (i.e. 3-1) times? It's certainly possible to add a UI option to calculate this for you, but you still need to provide parameters like how long each zone should turn on at a time, the total duration you want each zone to run etc. I mean, it basically saves a division calculation. And by 'additional configuration options' what exactly do you mean? What addition configuration parameters would you like to have?

Wetzel402, 2025-05-20:
I'm using the `use weather adjustment` feature. I want to cycle and soak 6 zones, each 5 minutes at at time for a total of 20 minutes of run time (not accounting for weather adjustments). I found that I had to set the `repeat every` field to 1 hour for it to get all 3 repeats, 4 total cycles. I initially expected I would want to set `repeat every` to 30 minutes as that is the expected run time per cycle (6 zones * 5 minutes each). 

As for configuration parameters, this would specifically be for a cycle and soak option. If the option is added to the UI there could be a field to configure cycle time as well as soak time. 

I hope that makes sense.

rayshobby, 2025-05-21:
OK. I see what's causing the issue. When using weather adjustment, the watering percentage may be higher than 100%. This can cause the total run time each cycle to be more than 30 minutes. The firmware in general can queue the next run while the previous is still running. However, it cannot queue the same zone if that zone hasn't finished running (i.e. already has an existing schedule). For example, if you set repeat every to 30 minutes, and if the watering percentage is 200%, what happens is that upon the second run, it's trying to schedule all 6 zones, but zones 4 to 6 haven't finished the previous run, so they cannot be scheduled again. On the contrary, setting the repeat every to 1 hour avoids the issue because by the time the program starts again, it has finished all previous zones (because the watering percentage can not be more than 200%). In fact, you can observe this in the Program Preview -- if you check Program Preview, settings 30 minutes would lead to some zones skipped in later runs; while setting 1 hour would avoid this issue.

Just adding a UI configuration option doesn't fix this problem because this is fundamentally due to the firmware not able to queue the same zone before it finishes the previous run, and it's also due to using watering percentage that's higher than 100% so it takes more than 30 minutes to go through each run. To fix this for good would require a pretty major firmware change (either allowing the same zone to be queued multiple times, or fundamentally change how the scheduler works). I will think about a work-around but it won't be an easy fix at the moment.

Wetzel402, 2025-05-23:
That makes sense. I appreciate the consideration. Hopefully there is an easily implementable scheduler change that can be made.

rayshobby, 2026-10-19:
Just a heads up that Cycle and Soak has been implemented in the upcoming firmware 2.2.1(7):
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/tree/dev/221_7
It allows you to set a maximum cycle time and minimum soak time for each zone individually, and will automatically split each zone's total water time into cycles no longer than the maximum cycle and gaps between cycles no shorter than the minimum soak. We are still testing firmware 2.2.1(7), it should be released by the end of next week.

rayshobby, 2026-11-06:
Per-zone Cycle and Soak is now implemented in master through #443 for the upcoming firmware 2.2.1(7) release. Closing as addressed. Thanks for the suggestion!

closed 2026-11-06 by rayshobby