OpenSprinkler/OpenSprinkler-Firmware #353: Only Some Email Notifications Working

Dillweed04, 2025-05-16:
I set up the email notifications through the gmail smtp. I only receive the email notifications for some of the selected notifications. I haven't tested all of them, but I don't get them for program starts or weather adjustments. 

I enabled the station notifications and those come through just fine.

rayshobby, 2025-05-19:
I've not received other reports of this issue. When you tested the program start notification, did you start a program manually, or was it an automatically started program (based on its schedule)? Weather adjustment notifications will only be sent when the  watering percentage changes -- if the watering percentage remains the same, it won't send any notification.

Dillweed04, 2025-05-19:
Ahh that must be it. I was starting the programs manually to test the notifications. I did have a scheduled program start yesterday, and I got the notification. 

For the watering percent changes does it not work when you have manual selected? I’m am running my own ETo script to account for multiple days between watering and setting the % through the api.

rayshobby, 2025-05-21:
Settings watering percentage via API will NOT trigger email notification, because this counts as changing an option value, it's not a change in the weather query result. The weather adjustment notification only applies to values changed due to weather query:
https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/main.cpp#L1071
It's pretty easy to modify the code to support notifications upon option value changes. But if you are using your own ETo script, you may just use the script to send notifications.

Dillweed04, 2025-05-23:
Sweet. Thank you for the help!

closed 2025-05-23 by Dillweed04