OpenSprinkler/OpenSprinkler-Firmware #354: Mexico time zone is incorrect

kareemk, 2025-05-19:
I'm setting my position to 15.68295,-96.56458 which should be Central Mexico Time -06:00 but the time zone is set to -05:00. This is likely due to the time zone database needing to be updated because Mexico recently in 2022 eliminated daylight savings

rayshobby, 2025-05-19:
The first thing you should check is whether the controller is receiving weather query result. At the homepage, swipe left to right to open the left side menu, then 'System Diagnostics'. See if there is any error message. If there is any error, then it's not receiving the time zone information, which explains the time zone error.

kareemk, 2025-05-19:
No error [image] and manually correcting the timezone by unsetting the location then resetting the location reverts the timezone back to the incorrect value

rayshobby, 2025-05-21:
So the issue turns out to be due to the OpenSprinkler Weather script needs to be updated to use the latest timezone database. I've created a fix:
https://github.com/OpenSprinkler/OpenSprinkler-Weather/commits/update-geotz/
and it will be deployed after some additional internal testing.

kareemk, 2025-05-21:
Amazing @rayshobby, thank you so much for the fix

rayshobby, 2025-05-27:
The new weather service has been deployed as of this morning and hopefully this fixes the issue. I've done a quick test and it seems working. You can re-open this issue if it still persists.

kareemk, 2025-05-27:
It's fixed! Thank you so much.