OpenSprinkler/OpenSprinkler-Firmware #372: [Bug] Can’t rename sprinklers

gimmel, 2025-08-13:
### Firmware Version

2.2.1 (1)

### Hardware Version

DC Version for PI

### Bug Description

In OpenSprinkler, my valves have names like S01, S02, etc. I can change the name in the web service, but no matter what I enter and save, the name is shown for a quarter of a second and the jumps back to S01, etc.

### Steps to Reproduce

1. Rename sprinkler
2. Save
3. See sprinkler name

rayshobby, 2025-08-19:
Can you please capture a complete video? As is the frame is cut off at the bottom and I can't see if you clicked on submit or not.

rayshobby, 2025-08-19:
I am trying to piece together this issue and the other issue you reported (#371). If I understand correctly, you can't save station names at all, not even in the same firmware session. In contrast, other settings like location, number of zones, you can change them and they stick in the same session but then are lost if you reboot your RPi or restart the firmware. Is that correct? If so, it's really bizarre because station names and other settings are stored in the same way on the flash driver, other than they are stored in different files. I can't think of a reason why some settings would at least survive in the same sessions while others (station names) can't even survive the same session.

rayshobby, 2025-08-19:
Station images are NOT stored by the firmware -- they are cloud-synced features and are stored on the cloud server so has nothing to do with the firmware or settings stored on the firmware side.

The fact the station name shows up for a fraction of a second then reverts back to default strongly suggests the changes were never saved to the disk file (specifically in `stns.dat`). This is because the UI briefly cached the station name change (hence it shows up briefly) until it retrieves the json data from the firmware (thus it reverts because the changes were never saved to disk). I am not sure what would be causing this, but all I can think of would be a weird write-protection / read-only setting on the folder.

arfrie22, 2025-08-20:
I am guessing that #371 was the same issue, if this still persists, you can leave a comment and I can reopen it.

closed 2025-08-20 by arfrie22