OpenSprinkler/OpenSprinkler-Firmware #378: [Bug] Relative start time doesn't change with sunrise

AustinHaigh, 2025-09-07:
### Firmware Version

2.2.1(1)

### Hardware Version

OSPi v2.0 (AC)

### Bug Description

I set the start time to be 60 minutes before sunrise, but now that we're getting into autumn I've noticed the actual start time isn't changing, causing the offset to drift. For example, one of my stations had a start time of exactly 4:55:51 AM on Sept 1st, Sept 4th, and Sept 7th. At my latitude, the start time should be changing by more than a minute each watering period, but even at the equator sunrise shifts a few seconds each day. I've checked that my location is set correctly.

The "Preview Programs" window does _not_ have this issue - it correctly shows the start time shifting later each day. The issue can only be seen by checking the logs.

### Steps to Reproduce

1. Set your location
2. Set up a watering program to have a start time relative to sunrise
3. After a few watering cycles, check the logs - the start time is the same each day

I've played with various settings since setting the start time for this watering program (eg. changing station run times, days interval, etc), but I can't remember exactly what changes were made or in what order.

rayshobby, 2025-09-07:
That is more likely due to the controller not getting correct sunrise time via the weather query. To check, at the homepage swipe left to right to open the side menu, then System Diagnosis. Check the weather request result. If there is any error that means it's not getting the weather request which contains sunrise and sunset times.

AustinHaigh, 2025-09-09:
The "Last Response" is "Weather Data Error," and `wterr = 1`.

I also confirmed that the firmware version is what I originally reported - 2.2.1(1).

To enable debug mode, do I build the firmware with ENABLE_DEBUG defined (ie. uncomment line 27 of defines.h)? Where can I find the logs?

austin2479, 2025-09-24:
Thanks. I'll send a PR to update the build instructions in `docs/local-installation.md`.

Unfortunately I'm still getting an error with `npm run build`. This is on a Raspberry Pi zero (32 bit), is there any chance you recently dropped support for 32bit ARM? The `npm run build` command works fine on my amd64 laptop.

```
pi@opensprinkler:~/weather $ npm run build

> os-weather-service@3.0.2 build
> node build.mjs

file:///home/austin/weather/build.mjs:35
await esbuild.build({
^^^^^

SyntaxError: Unexpected reserved word
    at Loader.moduleStrategy (internal/modules/esm/translators.js:133:18)
    at async link (internal/modules/esm/module_job.js:42:21)
npm ERR! code 1
npm ERR! path /home/austin/weather
npm ERR! command failed
npm ERR! command sh -c node build.mjs

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/austin/.npm/_logs/2025-09-24T23_22_01_464Z-debug.log

rayshobby, 2025-09-25:
Just paste the error to chatgpt, or any gen AI, or even google the error message, you will find out.

austin2479, 2025-09-25:
As far as I can tell, OpenSprinkler firmware no longer supports Raspbian Bullseye. Unfortunate if true, but I can upgrade. For what it's worth, the installation instructions I found here say it should work with Bullseye: https://openthings.freshdesk.com/support/solutions/articles/5000631599-installing-and-updating-the-unified-firmware

closed 2025-09-25 by rayshobby

austin2479, 2025-09-25:
I was referring to the OpenWeather-Weather script. Previous versions seemed to support Raspbian Bullseye (ie. no build or runtime errors), although I obviously wasn't able to get everything working.

rayshobby, 2025-09-26:
I think you are confusing OpenSprinkler-Weather with OpenSprinkler-Firmware. OpenSprinkler-Weather instructions never mentioned anything about Raspbian version. Regardless, as I said, paste the error message to any gen AI, or just google it, you will see the issue is your nodejs version is outdated. Either upgrade your nodejs, or modify the build script to remove the usage features not supported by your nodejs version.